openthread/ot-docs #24: Missing Border Agent docs for OTBR Docker

edmont, 2021-06-24:
The OTBR Docker guide lacks information about how to expose the IPv6 UDP port of the Border Agent to the host to be able to connect with the OT Commissioner: https://github.com/openthread/ot-docs/blob/main/site/en/guides/border-router/docker/run.md

Vyrastas, 2021-06-25:
@jwhui Any recommendation on who can help with this?

jwhui, 2021-06-25:
> @jwhui Any recommendation on who can help with this?

@edmont , is this something you can help contribute?

canisLupus1313, 2021-06-28:
Hi All,
there are two things that we should add at [linux setup](https://openthread.io/guides/border-router/docker) note about setting the ipv6 network for docker. The one from official docker docks [this link](https://docs.docker.com/config/daemon/ipv6/) should do the trick.
Add paragraph about checking if it was setup correctly eg.
```
$ ip addr
...
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:cd:7b:be:08 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
    inet6 2001:db8:1::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::42:cdff:fe7b:be08/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 fe80::1/64 scope link 
       valid_lft forever preferred_lft forever

...
```
NOTE the  inet6 2001:db8:1::1/64 scope global.

In the end the additional parameter for docker call should be added `-p <host port>:<otbr port>/udp`.
The only thing that I'm missing is how to determine OTBR commissioning port before running the container. 
@all Do You know how it's asigned?

jwhui, 2021-06-29:
> The only thing that I'm missing is how to determine OTBR commissioning port before running the container.
> Do You know how it's asigned?

@canisLupus1313 , by default the border agent port uses an ephemeral port. See:

1. https://github.com/openthread/openthread/pull/6450
2. https://github.com/openthread/ot-br-posix/pull/792

Maybe in the Docker case, we can configure a fixed port using `OPENTHREAD_CONFIG_BORDER_AGENT_UDP_PORT`.

@bukepo , thoughts?

bukepo, 2021-06-30:
How about adding a command line argument to otbr-agent for specifying the border agent port? We may then export the port at runtime by [--expose](https://docs.docker.com/engine/reference/run/#expose-incoming-ports).

Vyrastas, 2022-01-12:
Were any updates made for this issue or has it been resolved since?

Vyrastas, 2022-08-11:
Closing stale issue.

closed 2022-08-11 by Vyrastas