openthread/ot-docs #32: Information missing in codelab

OLmmyT, 2021-07-21:
An information is missing in this codelab: https://openthread.io/codelabs/openthread-hardware#4

It is not indicated that you should use the "nRF USB port". If you don't, you don't get the prompt when you call the screen command.

jwhui, 2021-07-21:
@OLmmyT , the codelab currently uses the `UART_trans` flag in the build command. When using `UART_trans`, then the instructions are as stated in the codelab:
```
Attach the USB cable to the Micro-USB port next to the external power pin on the nRF52840 board
```

The nRF USB port only applies when using the `USB_trans` flag.

OLmmyT, 2021-07-22:
Ok thank you Jonathan.

I have USB_trans flag because I'm following another codelab (https://openthread.io/codelabs/openthread-border-router#3) which says to follow the instructions on this page (https://openthread.io/codelabs/openthread-hardware#4). 

> Follow step 5 of this codelab to build and flash a nRF52840 CLI end device. But instead of having OT_COMMISSIONER and OT_JOINER enabled, the CLI node requires OT_SRP_CLIENT and OT_ECDSA features:
> 
> 
> `$ script/build nrf52840 USB_trans -DOT_SRP_CLIENT=ON -DOT_ECDSA=ON`

wgtdkp, 2021-07-22:
@jwhui https://openthread.io/codelabs/openthread-hardware uses both `USB_trans` and `UART_trans` in different steps, should we change to always use `UART_trans`.

jwhui, 2021-07-22:
> @jwhui https://openthread.io/codelabs/openthread-hardware uses both `USB_trans` and `UART_trans` in different steps, should we change to always use `UART_trans`?

I agree.

jwhui, 2021-07-22:
@OLmmyT , thanks for raising this issue. I submitted a proposed change in #36 .

PR #36 (2021-07-23): [codelabs] update hw codelabs to always use USB instead of UART
Fixes #32
closed 2021-07-23 by jwhui