openthread/ot-docs #33: Questions about codelab

OLmmyT, 2021-07-22:
I have completed this codelab: https://openthread.io/codelabs/openthread-border-router#0

Can you please point me:
- How to see the list of end devices in the Thread network? Shall I use "sudo service otbr-agent status" on the border router?
- How to disconnect the end device (I want to redo the operation but now with the open thread commissioning app).

Thank you!
Regards

wgtdkp, 2021-07-22:
> How to see the list of end devices in the Thread network? Shall I use "sudo service otbr-agent status" on the border router?

You can view the network topology with OTBR web: https://openthread.io/guides/border-router/web-gui.

> How to disconnect the end device (I want to redo the operation but now with the open thread commissioning app).

you can run command `factoryreset` on the end device to factory reset it.

OLmmyT, 2021-07-22:
Thanks.
In the meantime I have been able to use the Border Router Web GUI:
In Android Service Browser, I have clicked on _meshcop._udp service, copied the IP address (192.168.43.15 in my case) and pasted it in Chrome (still on Android phone). I'm able to display the Border Router Web GUI and to see the topology.

> you can run command factoryreset on the end device to factory reset it.
I have used "thread stop". If I use the command "factoryreset", will the end device forget all the dataset corresponding to the border router?

wgtdkp, 2021-07-22:
> I have used "thread stop". If I use the command "factoryreset", will the end device forget all the dataset corresponding to the border router?

Yes, it is a factory reset.

closed 2021-07-22 by OLmmyT