openthread/ot-docs #39: Cannot check state of a node

jagra26, 2021-08-10:
I cannot check the node 2 state in this codelab: https://openthread.io/codelabs/openthread-simulation#4 

[image]

prompt dont shows ">"

Vyrastas, 2021-08-10:
Hi @jagra26, the state command should be run in a separate terminal window using `ot-ctl`, as per **3. Use ot-ctl to join the network** on that same page.  Have you tried that?

jagra26, 2021-08-10:
i have three terminals, one per node. i do have a fourth for run the state command?

jagra26, 2021-08-10:
first terminal: 
[image]

second terminal: 
[image]

third terminal:
[image]

Vyrastas, 2021-08-10:
You run it in the third terminal as described in the Codelab.

jagra26, 2021-08-10:
the codelab says: 
[image]

[image]
the node 2 is the second terminal not the third

jagra26, 2021-08-10:
The text of codelab are ambiguous. Run the commands in third terminal works, but not clear. Thanks for the help

Vyrastas, 2021-08-10:
The text of the codelab is clear, you left out the part just prior to checking the state, which explicitly states:

[image]

Vyrastas, 2021-08-10:
Actually, looking at it again, I see what you mean and how it could be confusing.  In the codelab when you're supposed to do commands in other windows it will say "On node 1".  It doesn't here because the user should just continue with the same terminal unless told to switch.  I can make a tweak so it's clear you're not supposed to switch windows and that the window where you started node 2 cannot be used for commands.

closed 2021-08-11 by jwhui