openthread/ot-docs #43: otUdpBind example usage need to be updated

rretanubun, 2021-09-18:
[OpenThread PR#6761][1] added a new arg for `otUdpBind(..., otNetifIdentifier aNetif)`. 
[The example][2] in the codelab is still using the old API.

[1]: https://github.com/openthread/openthread/pull/6761
[2]: https://openthread.io/codelabs/openthread-apis#6

When the example is build, the compile error out like this:
```
../openthread/examples/apps/cli/main.c:319:5: error: too few arguments to function 'otUdpBind'
  319 |     otUdpBind(aInstance, &sUdpSocket, &listenSockAddr);
      |     ^~~~~~~~~
In file included from ../openthread/examples/apps/cli/main.c:46:
../openthread/include/openthread/udp.h:209:9: note: declared here
  209 | otError otUdpBind(otInstance *aInstance, otUdpSocket *aSocket, const otSockAddr *aSockName, otNetifIdentifier aNetif);
      |         ^~~~~~~~~
```

Not sure if `OT_NETIF_UNSPECIFIED` or `OT_NETIF_THREAD` is the correct answer for this example code.

Vyrastas, 2021-09-23:
Thanks for reporting this.

@jwhui do you know what value should be used in the codelab for this new parameter?

jwhui, 2021-09-27:
> @jwhui do you know what value should be used in the codelab for this new parameter?

We should use `OT_NETIF_THREAD`.

Vyrastas, 2021-09-29:
@sarah-iot Can you make this update, please.