openthread/ot-docs #61: simulation codelab "commissioner joiner add * JO1NME" didn't work

hptechdesign, 2021-12-11:
When following the tutorial here:
https://openthread.io/codelabs/openthread-simulation#3

I got stuck trying to execute the command:
`> commissioner joiner add * JO1NME
Error 7: InvalidArgs`

Changing to this worked:
`> commissioner joiner add * AAAABBBB
Done`

But I don't know why

wgtdkp, 2021-12-12:
The Joiner password is actually `J01NME` - it's zero but not uppercase `O`.
This is the requirement of Thread spec that `I`, `O`, `Q` and `Z` should be avoid for readability (to avoid confusion with numbers).

wgtdkp, 2021-12-12:
@jwhui @sarah-iot Should we document it clear what characters are accepted for the joiner password?

hptechdesign, 2021-12-12:
Ah, thanks, I can't believe I missed that..
If the InvalidArgs error was caused by me using illegal characters then it might help to document which are illegal at the point of first use. It would help if the "InvalidArgs" error message was given an "IllegalChars" suffix.
In any case, that's clear to me now, thanks for your help

jwhui, 2021-12-13:
> @jwhui @sarah-iot Should we document it clear what characters are accepted for the joiner password? Or use an example password which is not misleading (`J01NME` looks like `JOINME`...).

I think we can put a note on the specific alphabet that is accepted. That said, the font on the codelab clearly shows the difference between `0` and `O`, the former having the diagonal slash.

PR #62 (2021-12-14): Issue 61 - Add Joiner Requirements to Codelabs
Fixes #61
closed 2021-12-14 by jwhui