openthread/ot-docs #67: Codelab 'Simulating a Thread network using OpenThread in Docker': unable to run ot-daemon

j-noel-longhand, 2022-01-10:
The Docker Thread sim Codelab is not currently working. 

The problems begin where you start the ot-daemon in container #2 (https://openthread.io/codelabs/openthread-simulation#4).

Initially, the daemon was unable to find the `ipset` command (apparently this is missing from the Docker image). 

Once `ipset` is installed, however, it appears to fail because no ip sets exist:

```
root@162dbab6802a:~/src/openthread# ./output/posix/bin/ot-daemon -v 'spinel+hdlc+forkpty://output/simulation/bin/ot-rcp?forkpty-arg=2'
./output/posix/bin/ot-daemon[170]: Running OPENTHREAD/g442b7e82f; POSIX; Jan  7 2022 05:52:28
./output/posix/bin/ot-daemon[170]: Thread version: 3
./output/posix/bin/ot-daemon[170]: Thread interface: wpan0
./output/posix/bin/ot-daemon[170]: RCP version: OPENTHREAD/g442b7e82f; SIMULATION; Jan  7 2022 05:50:28
ipset v6.34: Kernel support protocol versions 6-7 while userspace supports protocol versions 6-6
The set with the given name does not exist
```

The daemon never returns control to the terminal, which prevents one from issuing any further commands to the OT CLI in that container instance.

Looks like:
1) the Docker image needs to be updated to include `/sbin/ipset`, or the Codelab should be updated to tell the user to install it, and
2) either the Docker image needs to be updated with defined ip sets, or instructions need to be added to the Codelab explaining how to create the required ip sets.

Experienced the same issue on two different Debian machines, one running the latest vanilla Debian Stretch.

wgtdkp, 2022-01-11:
@superwhd This seems related to the latest ingress filter feature, will you take a look?

superwhd, 2022-01-11:
Sent a fix: https://github.com/openthread/openthread/pull/7305. This issue was caused by a feature `firewall` that shouldn't have been enabled on ot-daemon. Although enabling it won't break the ot-daemon's behavior, we'd better disable it to avoid such confusing logs.

shuhain, 2022-01-12:
sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM1?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[3281]: Running OPENTHREAD/thread-reference-20200818-1443-g55af6ce2c; POSIX; Jan 11 2022 18:04:30
./build/posix/src/posix/ot-daemon[3281]: Thread version: 3
./build/posix/src/posix/ot-daemon[3281]: Thread interface: wpan1
./build/posix/src/posix/ot-daemon[3281]: RCP version: OPENTHREAD/thread-reference-20200818-1436-g442b7e82f; NRF52840; Jan 10 2022 22:39:12
sh: 1: ipset: not found

superwhd, 2022-01-12:
The fix PR has been merged on main branch. I tried the codelab locally and the `ipset` error message doesn't show. I'm closing this bug.

@j-noel-longhand @shuhain Please update your repository/docker image. Let me know if you still see the error.

closed 2022-01-12 by superwhd