openthread/ot-docs #68: No ping6 in codelab_otsim

robertcragie, 2022-01-11:
When following https://openthread.io/codelabs/openthread-simulation#4 step 5, I get the following:
```
root@a761053b08c5:~/src/openthread# ping6 -c 4 fdd5:a079:4a43:2154:7d6a:709b:42d:a16d
bash: ping6: command not found
```
If I manually install the `iputils-ping` package, it works OK, so it looks like `ping6` is missing from the Docker image.

It also says:
> on your host machine's command line, ping Node 1

whereas it makes more sense to me to say something like:
> on the container's command line, ping Node 1

jwhui, 2022-01-12:
@robertcragie , thanks for raising this issue!

While we're updating the Dockerfile and codelab, I'm wondering if we should consolidate [openthread/codelab_sim](https://github.com/openthread/openthread/blob/main/etc/docker/codelab_otsim/Dockerfile) into [openthread/environment](https://github.com/openthread/openthread/blob/main/etc/docker/environment/Dockerfile). Doing so will take advantage of CMake/Ninja build and allow us to maintain a single Dockerfile long term. @Vyrastas , thoughts?

Vyrastas, 2022-01-13:
> @robertcragie , thanks for raising this issue!
>
> While we're updating the Dockerfile and codelab, I'm wondering if we should consolidate [...]

Yeah I think that makes sense at this point.  Let's resolve this with https://github.com/openthread/openthread/pull/7313 to get it working in the short term, and do this as a separate issue.  I'll create one for it.

Vyrastas, 2022-01-13:
Created https://github.com/openthread/openthread/issues/7314 for consolidating the Dockerfiles.

jwhui, 2022-01-13:
Resolved by https://github.com/openthread/openthread/pull/7313 and #70.

closed 2022-01-13 by jwhui