openthread/ot-docs #73: Missing old Raspberry Pi SoftAP configuration

jumpingjackgazz, 2022-02-02:
Hi there,

Back ago, I've followed on opentherad.io a guide to setup a SoftAP on the Raspberry Pi and configuration of all the services (dnsmasq, hostapd, tayga, NAT, ip fwd, etc...) to have a running setup to use along with Thread 1.1 external commissioning App.
Today I'd like to rebuild the same setup with a up to date OTBR and RCP code but this part is missing in the docs. 
Commit 6a8ae56e4facc791552cc84807b148bc82504391 suggests it has been done on purpose.

What is the reason behind that ?
BTW, today's doc suggests RPi is still providing an AP.

[image]
 
Thx and Regards

jwhui, 2022-02-02:
@jumpingjackgazz , yes, we intentionally removed the Raspberry Pi Wi-Fi AP setup documentation because it was difficult for us to maintain and track changes across the various Raspberry Pi releases.

You may find the official [Raspberry Pi Documentation](https://www.raspberrypi.com/documentation/computers/configuration.html) helpful, which does cover networking configuration.

jumpingjackgazz, 2022-02-03:
Ok thanks for your quick answer !

jumpingjackgazz, 2022-02-03:
BTW, I have cloned the ot-docs repo and checkout commits prior to this commit.
There is no trace of file ../../guides/border-router/access-point.md

Any means to get back this old doc ?
Usefull to get the main RPi configuration topics

Thx and regards

Vyrastas, 2022-02-03:
Hi @jumpingjackgazz that page was never sourced on GitHub, it was removed from openthread.io before we migrated the rest of the border router content over.

There is a snapshot available on the Wayback Machine, though, let me know if that helps: https://web.archive.org/web/20210417113340/https://openthread.io/guides/border-router/access-point

jumpingjackgazz, 2022-02-04:
Hi @Vyrastas, thx a lot for the link. That's helping indeed !

Vyrastas, 2022-02-16:
We don't plan on bringing back this documentation, so hopefully what I provided solved your issue @jumpingjackgazz .  Closing.

closed 2022-02-16 by Vyrastas