openthread/ot-docs #76: Silicon Labs EFR32 Codelab Updates

sarah-iot, 2022-02-10:
Hi @silabs-mithil,
I was just going through our documentation and noticed that the [EFR32 Codelab](https://openthread.io/codelabs/silabs-openthread-hardware#6) is referencing an older Docker tag: `siliconlabsinc/openthread-border-router:gsdk-3.2.0`

I see that there are some newer images: https://hub.docker.com/r/siliconlabsinc/openthread-border-router/tags

Would it be possible to update this Codelab to a more recent Thread 1.2 version? Is it just a matter of updating the tag, or does the Codelab need additional updates too?

Thanks so much, and please let me know if there's anything I can do to help. I'm happy to make any updates required.
Sarah
@Vyrastas

sarah-iot, 2022-02-10:
@silabs-mithil - Thanks for getting back to me. Sure, no problem; I've already started working on a branch. I just have one question: would it be alright if I added some of the Docker content from the PDF? I can add more clarity to the Codelab, and I can also try to re-work the Docker section just a tiny bit to allow for new, future GSDK tags (so we won't need to update it every time - the idea is to make the documentation less version specific if possible).

I can start a draft and send it out for a PR for you to review...I just wanted to run it by you first before I got started.

I realize that things might change with new versions/tags, so if it makes sense to just update the Codelab to `gsdk-4.0.1`, I can do that too.

Thanks,
Sarah

silabs-mithil, 2022-02-11:
@sarah-iot , sounds good to me.

silabs-mithil, 2022-02-14:
https://github.com/openthread/ot-docs/pull/77 resolves this issue.

closed 2022-02-14 by silabs-mithil