openthread/ot-docs #88: Unable to quit ot-ctl by using Ctrl+D

jcdong98, 2022-06-22:
In the chapter *Manage the network with OpenThread Daemon* of the codelab *Simulating a Thread network with OpenThread* (site/en/codelabs/openthread-simulation-posix/index.lab.md), the document states that quit ot-ctl by using Ctrl+D, which did not work on my macOS 12.4, with the current latest Thread version 4. In addition, ot-daemon will not terminate as instructed when trying to use `exit` command to quit ot-ctl. I guess that the mechanism may change after the version upgrades.

wgtdkp, 2022-06-22:
I can confirm that `ot-ctl exit` won't terminate `ot-daemon` now. @jcdong98 Will you help update the doc?

jcdong98, 2022-06-22:
> I can confirm that `ot-ctl exit` won't terminate `ot-daemon` now. @jcdong98 Would you help update the doc?

OK, I will make a PR later.

bukepo, 2022-06-23:
I think this is only a limitation on macOS.

jcdong98, 2022-06-23:
> I think this is only a limitation on macOS.

@bukepo I have re-run the procedure on Ubuntu 22.04 with the latest OpenThread. The result on Linux indicates the behavior of Ctrl+D in ot-ctl is consistent with that of `exit` command. Both of them leads to termination of ot-ctl, however, will **NOT** terminate ot-daemon. The latter still conflicts with *Warning* section in the chapter *Manage the network with OpenThread Daemon*.

bukepo, 2022-06-24:
Yes, ctrl+d is intended to be consistent with the `exit` command. And both only exit ot-ctl instead of `ot-daemon`.