openthread/ot-docs #144: Requirement for  Raspberry Pi OS with Desktop?

olavt, 2023-12-04:
Related to this page: https://openthread.io/guides/border-router/docker

I think the requirement for using  Raspberry Pi OS with Desktop is a bit confusing.

"You cannot use the Lite version, as you need to access the OTBR Web GUI in a web browser."

In my experience it's possible to open the OTBR Web GUI in a web browser running on another computer.

Vyrastas, 2023-12-04:
@jwhui do you know who can address this question, and if we need to update the docs?  We haven't updated them in a while so don't know if this needs to be changed/removed.

jwhui, 2024-06-06:
Yeah, I think the point here is that the "Lite" version still allows a web browser to view, but the web browser has to be on a different computer.

Vyrastas, 2024-06-07:
Thanks, we can add a note to that effect on the docs.  @jrhodie can you make an update for this?  

https://github.com/openthread/ot-docs/blob/main/site/en/guides/border-router/docker/index.md

jrhodie, 2024-06-07:
will do, thanks

closed 2024-06-10 by jwhui

jrhodie, 2024-06-10:
note has been added and approved, and merged.  Will run copybara vs main to have it appear.

jrhodie, 2024-06-10:
closing this issue