openthread/ot-docs #163: TCP ports and how to persist data @ Dokcer

snakuzzo, 2024-07-22:
I'm running OTBR on Docker.

Is it possible to specify:

- Every TCP port used. On doc there is just TCP port related to the webgui but, for example, for some integrations we need also 8081 (used by OTBR for websocket protocol). Are there any other used TCP ports?
- If I destroy my container I lose every configuration about thread network. What are directories to bind to make data persistent ?

Thank you

KC-inDomus, 2024-07-27:
can't believe this google project is so badly mantained and documented.

jwhui, 2024-07-31:
> Every TCP port used. On doc there is just TCP port related to the webgui but, for example, for some integrations we need also 8081 (used by OTBR for websocket protocol). Are there any other used TCP ports?

It's just the two ports you identified - one for the REST API and another for the web UI.

> If I destroy my container I lose every configuration about thread network. What are directories to bind to make data persistent ?

The current OTBR Docker image is meant for development, not production, and does not save state.

You can make `/var/lib/thread` a Docker volume. More here: https://github.com/orgs/openthread/discussions/10311#discussioncomment-9754171

jwhui, 2024-07-31:
> can't believe this google project is so badly mantained and documented.

OpenThread is a community supported project. Anyone is free to make contributions.

KC-inDomus, 2024-07-31:
> > can't believe this google project is so badly mantained and documented.
> 
> 
> 
> OpenThread is a community supported project. Anyone is free to make contributions.

I would do it, if I were able.
So, as a simple user, I feel uncomfortable with so little support.

jwhui, 2024-07-31:
> So, as a simple user, I feel uncomfortable with so little support.

You can start by submitting a PR that helps address any issues you've run into.