openthread/ot-docs #165: OTBR-Docker in combination with a network device

hammy1988, 2024-08-20:
Hello,

I have a question regarding the OTBR docker.

On the internet I find many instructions how to realize Matter-over-Thread with a SLZB-06 in Home Assistant. Unfortunately, I run Home Assistant in a Docker, so I cannot use the OTBR integration via the add-on.
Fortunately, there is the Docker image provided here :)

How is it possible to add the SLZB-06 as a network device (like in the Home Assistant add-on?

I would appreciate an answer.

Best regards
Hammy

jwhui, 2024-08-27:
I would reach out to the Home Assistant community of SLZB-06 vendor directly.

OpenThread is designed to interface with a radio co-processor (RCP) using a standard serial protocol. So the specific serial port settings depends on the specific firmware that the RCP is running.

mattaustin, 2024-11-06:
Hi @hammy1988 , did you ever work this out? I have the same question.

zerpex, 2024-12-14:
You have it all on the help:

`Usage: otbr-agent [-I interfaceName] [-B backboneIfName] [-d DEBUG_LEVEL] [-v] [-s] [--auto-attach[=0/1]] RADIO_URL [RADIO_URL]
     -I, --thread-ifname    Name of the Thread network interface (default: wpan0).
     -B, --backbone-ifname  Name of the backbone network interfaces (can be specified multiple times).
     -d, --debug-level      The log level (EMERG=0, ALERT=1, CRIT=2, ERR=3, WARNING=4, NOTICE=5, INFO=6, DEBUG=7).
     -v, --verbose          Enable verbose logging.
     -s, --syslog-disable   Disable syslog and print to standard out.
     -h, --help             Show this help text.
     -V, --version          Print the application's version and exit.
     --radio-version        Print the radio coprocessor version and exit.
     --auto-attach          Whether or not to automatically attach to the saved network (default: 1).
     --rest-listen-address  Network address to listen on for the REST API (default: [::]).
     --rest-listen-port     Network port to listen on for the REST API (default: 8081).

RadioURL:
Radio Url format:    {Protocol}://${PATH_TO_DEVICE}?${Parameters}

Protocol=[spinel+hdlc*]           Specify the Spinel interface as the Spinel HDLC interface
    forkpty-arg[=argument string]  Command line arguments for subprocess, can be repeated.
    spinel+hdlc+uart://${PATH_TO_UART_DEVICE}?${Parameters} for real uart device
    spinel+hdlc+forkpty://${PATH_TO_UART_DEVICE}?${Parameters} for forking a pty subprocess.
Parameters:
    uart-parity[=even|odd]         Uart parity config, optional.
    uart-stop[=number-of-bits]     Uart stop bit, default is 1.
    uart-baudrate[=baudrate]       Uart baud rate, default is 115200.
    uart-flow-control              Enable flow control, disabled by default.
    uart-init-deassert             Deassert lines on init when flow control is disabled.
    uart-reset                     Reset connection after hard resetting RCP(USB CDC ACM).

    region[=region-code]          Set the radio's region code. The region code must be an
                                  ISO 3166 alpha-2 code.
    cca-threshold[=dbm]           Set the radio's CCA ED threshold in dBm measured at antenna connector.
    enable-coex[=1|0]             If not specified, RCP coex operates with its default configuration.
                                  Disable coex with 0, and enable it with other values.
    fem-lnagain[=dbm]             Set the Rx LNA gain in dBm of the external FEM.
    no-reset                      Do not send Spinel reset command to RCP on initialization.
    skip-rcp-compatibility-check  Skip checking RCP API version and capabilities during initialization.
    bus-latency[=usec]            Communication latency in usec, default is 0.`

You just have to chain params with an "&" between them.

example:
`--radio-url spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=460800&uart-stop=16`

Some more generic params are loaded from environment variables such as FIREWALL or WEB_GUI.

closed 2025-04-02 by jwhui