openthread/ot-docs #183: Review ESP Thread Border Router instructions

Vyrastas, 2025-06-20:
We got some feedback through openthread.io on the ESP Thread Border Router guide:

> it looks like there are mistakes on this page. ./install.sh should really be .\install.bat . ./export.sh should really be . .\export.bat

Guide in question: https://github.com/openthread/ot-docs/blob/main/site/en/guides/border-router/espressif-esp32.md

Vyrastas, 2025-06-20:
@chshu Can you take a look at this?

chshu, 2025-06-23:
@Vyrastas The default commands `./install.sh` and `./export.sh` are for Linux and macOS. For Windows, should use `.\install.bat` and `.\export.bat` instead.

We've included an [official installation guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/get-started/index.html#installation-step-by-step) in the docs for new users who may not be familiar with the IDF environment.

Vyrastas, 2025-06-24:
@jrhodie Can you make the update mentioned in the comment above?  Then we can import into openthread.io.

jrhodie, 2025-06-24:
https://github.com/openthread/ot-docs/pull/185

jrhodie, 2025-06-25:
fixed in https://critique.corp.google.com/cl/775370381

closed 2025-06-25 by jrhodie